KumarRobotics/ublox #222: build fail in foxy

BartVanderHaagen, 2023-07-12:
**Describe the bug**
build fail 

**To Reproduce**
On foxy : 
bart@ubuntu:~$ mkdir gps_ws
bart@ubuntu:~$ cd gps_ws/
bart@ubuntu:~/gps_ws$ mkdir src
bart@ubuntu:~/gps_ws$ cd src
bart@ubuntu:~/gps_ws/src$ git clone -b ros2 https://github.com/KumarRobotics/ublox.git
bart@ubuntu:~/gps_ws/src$ cd ..
bart@ubuntu:~/gps_ws$ colcon build 
Starting >>> ublox_serialization
Finished <<< ublox_serialization [6.20s]                  
Starting >>> ublox_msgs
--- stderr: ublox_msgs                           
CMake Error at CMakeLists.txt:107 (rosidl_get_typesupport_target):
  Unknown CMake command "rosidl_get_typesupport_target".

**Expected behavior**
the build to complete without this error

**Desktop (please complete the following information):**
 - OS Ubuntu 20.04
 - ROS Version ROS2 Foxy]

gokulp01, 2023-07-13:
Hi, this is a persisting issue. You should try cloning the `foxy-devel` branch. It works perfectly without any errors.

BartVanderHaagen, 2023-07-14:
I found the brach and it indeed compiled perfect!