KumarRobotics/ublox #224: How solve this error? plz.... quick answer

sonny0929, 2023-07-20:
Starting >>> ublox_serialization
Finished <<< ublox_serialization [0.10s]
Starting >>> ublox_msgs
Finished <<< ublox_msgs [1.69s]                     
Starting >>> ublox_gps
--- stderr: ublox_gps                         
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find asio (missing: ASIO_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  cmake/Findasio.cmake:4 (find_package_handle_standard_args)
  CMakeLists.txt:16 (find_package)

---
Failed   <<< ublox_gps [0.16s, exited with code 1]

Summary: 2 packages finished [2.09s]
  1 package failed: ublox_gps
  1 package had stderr output: ublox_gps
  1 package not processed

sonny0929, 2023-07-20:
i'm korean student and preparing an autonomous ship comepetition 
please give me answer ㅠㅠ

gokulp01, 2023-07-25:
Can you try `sudo apt-get install -y libasio-dev` and build again

BillyMazotti, 2025-02-22:
This worked for me on raspberry pi 5 running ubuntu 24.04 and building with colcon for ros2 branch