KumarRobotics/ublox #226: The problem of the input buffer overflow

yulan0215, 2023-08-02:
**Describe the bug**
When I ran the Ublox ROS driver, after several minutes. The terminal showed: U-blox ASIO input buffer overflow, dropping 8192 bytes.   Do you know how to solve this problem?

**To Reproduce**
Steps to reproduce the behavior:
1. I just ran the roslaunch file to start the U-blox ROS driver.

**Expected behavior**
Without the error in the terminal

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - ROS Version [e.g. ROS1 Melodic]
 - Jetson Nano

gokulp01, 2023-08-03:
What is the receiver you are using? Can you try updating your config file to receive just these and see if it works fine?

```
    gnss:
      glonass: false
      beidou: false
      gps: true
      qzss: false
      galileo: false
      imes: false
```

yulan0215, 2023-08-03:
> What is the receiver you are using? Can you try updating your config file to receive just these and see if it works fine?
> 
> ```
>     gnss:
>       glonass: false
>       beidou: false
>       gps: true
>       qzss: false
>       galileo: false
>       imes: false
> ```
Many thanks for your quick reply! I used Ublox ZED-F9P.

yulan0215, 2023-08-03:
> What is the receiver you are using? Can you try updating your config file to receive just these and see if it works fine?
> 
> ```
>     gnss:
>       glonass: false
>       beidou: false
>       gps: true
>       qzss: false
>       galileo: false
>       imes: false
> ```

I am sorry I still had this problem, therefore, I uploaded my configureation file here:
[image]

And the measure rate was 8 Hz(I would like to have quicker RTK solution)

Do you have any suggested measure rate to overcome this problem? Thx

gokulp01, 2023-08-03:
I see, I am a little unsure why it is occurring -- it could be because of Firmware 9. I have a repository that has ported this codebase with necessary changes for it to work with Ublox ZED-F9P (but on ROS2) Linking it here in case it helps you: https://github.com/gokulp01/ros2-ublox-zedf9p. 

Nevertheless, this is the config file that works for me. 
This might be trivial, but I would suggest also checking your launch file if it is pointing to the correct config file. 

```
ublox_gps_node:
  ros__parameters:
    debug: 0                    # Range 0-4 (0 means no debug statements will print)
    device: /dev/ttyACM0
    frame_id: gps
    uart1:
      baudrate: 460800
    gnss:
      glonass: false
      beidou: false
      gps: true
      qzss: false
      galileo: false
      imes: false
    # TMODE3 Config
    tmode3: 1                   # Survey-In Mode
    sv_in:
      reset: True               # True: disables and re-enables survey-in (resets)
                                # False: Disables survey-in only if TMODE3 is
                                # disabled
      min_dur: 300              # Survey-In Minimum Duration [s]
      acc_lim: 3.0              # Survey-In Accuracy Limit [m]
 
    inf:
      all: true                   # Whether to display all INF messages in console
 
    publish:
      all: false
      nav:
        all: true
        relposned: true
        posllh: true
        posecef: true
```

Also, I believe the update rate for ZEDF9P is this:
[image]

I would also suggest trying to keep it less than 5Hz and check if it works in your case.

yulan0215, 2023-08-03:
> I see, I am a little unsure why it is occurring -- it could be because of Firmware 9. I have a repository that has ported this codebase with necessary changes for it to work with Ublox ZED-F9P (but on ROS2) Linking it here in case it helps you: https://github.com/gokulp01/ros2-ublox-zedf9p.
> 
> Nevertheless, this is the config file that works for me. (I would suggest following the same format mentioned in the repository) This might be trivial, but I would suggest also checking your launch file if it is pointing to the correct config file.
> 
> ```
> ublox_gps_node:
>   ros__parameters:
>     debug: 0                    # Range 0-4 (0 means no debug statements will print)
>     device: /dev/ttyACM0
>     frame_id: gps
>     uart1:
>       baudrate: 460800
>     gnss:
>       glonass: false
>       beidou: false
>       gps: true
>       qzss: false
>       galileo: false
>       imes: false
>     # TMODE3 Config
>     tmode3: 1                   # Survey-In Mode
>     sv_in:
>       reset: True               # True: disables and re-enables survey-in (resets)
>                                 # False: Disables survey-in only if TMODE3 is
>                                 # disabled
>       min_dur: 300              # Survey-In Minimum Duration [s]
>       acc_lim: 3.0              # Survey-In Accuracy Limit [m]
>  
>     inf:
>       all: true                   # Whether to display all INF messages in console
>  
>     publish:
>       all: false
>       nav:
>         all: true
>         relposned: true
>         posllh: true
>         posecef: true
> ```
> 
> Also, I believe the update rate for ZEDF9P is this: [image]
> 
> I would also suggest trying to keep it less than 5Hz and check if it works in your case.
> 
> Lastly, I would also check if the receiver is working using u-center: https://www.u-blox.com/en/product/u-center

Many thanks and I will test later!

yulan0215, 2023-08-04:
> I see, I am a little unsure why it is occurring -- it could be because of Firmware 9. I have a repository that has ported this codebase with necessary changes for it to work with Ublox ZED-F9P (but [...]

Hi, I decreased the rate below 5 Hz, however, I still had this problem...
I set it to be 5Hz(I set it to be higher because I would like to get the RTK solution as soon as possible)

The baudrate I set it to be 460800... 

The only difference is here:

# TMODE3 Config
    tmode3: 1                   # Survey-In Mode
    sv_in:
      reset: True               # True: disables and re-enables survey-in (resets)
                                # False: Disables survey-in only if TMODE3 is
                                # disabled
      min_dur: 300              # Survey-In Minimum Duration [s]
      acc_lim: 3.0              # Survey-In Accuracy Limit [m]
 
    inf:
      all: true                   # Whether to display all INF messages in console
I did not config this part because my Ublox was Rover.
Another problem was that it would take a little bit long time to get the fix mode of RTK... Will take several to 10 min, do you have any idea about this? Thx~