KumarRobotics/ublox #236: Inconsistent heading data across ros distros zed-f9p

xaru8145, 2024-01-29:
Hi,

I have a setup that consists in using 2 u-blox zed-9fp receivers from the [simpleRTK2B Heading](https://www.ardusimple.com/product/simplertk2b-heading-basic-starter-kit-ip67/) kit. Both receivers are flashed with firmware version 1.30.

One receiver is configured as the moving base and receives ntrip corrections while the other is configured as the rover and receives data from the moving base in order to calculate the heading.

I am seeing an issue where the heading data that comes from the rover is different in I ros1 compared to ros2. In ros1 I use the tag 1.4.1 and the IMU data from the topic /rover/navheading is consistent with ENU frame according to ROS convention (REP-103). However, it seems that in the ros2 branch the topic /rover/navheading is not calculated correctly and the values read from the device in the relPosHeading field of NavRELPOSNED9.msg does not match in ros1 and ros2. 

Has someone experienced a similar issue? I can provide more documentation and context if necessary.

Thanks in advance!

bnbhat, 2024-01-31:
Hey!

I am also trying to do the same with a pair of **u-blox zed-f9p** and **u-blox zed-f9h** modules.
I set them up in u-center and it is working. But, I'm having difficulties configuring them with this driver as a moving base and a rover. 

Would you mind sharing your configuration files? It would be a great help to me!

Thanks in advance!

xaru8145, 2024-01-31:
Hi  @bnbhat !

If you see your data correctly in the u-center, it means that the receiver is well configured. 

If they are not when you launch the driver, I am guessing that you are trying to override the configuration through setting ros parameters through the driver. In order to avoid this, try setting the `config_on_startup` parameter to `false`. 

You just then need to specify the right port, the baudrate and which topics you want to publish, like on the example [config file](https://github.com/KumarRobotics/ublox/blob/master/ublox_gps/config/zed_f9p.yaml) provided in this package.

bnbhat, 2024-02-01:
Thank you so much! @xaru8145 

As you suggested, I changed `config_on_startup` to `false` and it is working now!

Yes, I am using ROS 2 Humble. For now, it seems the heading is the same as what I can see in the u-center.

<div style="display: flex; justify-content: space-between;">
  [image]
  [image]
</div>
  
   
Here you can find my configurations [https://github.com/Lemvos/ublox_ros2_driver.git](url)
u-center config for f9p as a moving base [https://gist.github.com/bnbhat/c27f6d7f87c88cf15f6a0633b90dcc80](url)

Let me know if you need any information about my setup.

xaru8145, 2024-02-05:
Thanks @bnbhat ! I tried using 2.3.0 release in humble and it works just fine. It seems that porting the 2.3.0 release from humble to foxy and making it compile is not enough. There must be something related to differences in how these 2 ROS distros serialize messages. Best thing is to forget about foxy and move to humble

closed 2024-02-13 by xaru8145

ciniminis1, 2024-10-02:
Hi @xaru8145 , I plan to use the same setup to get a heading information. Please, is the heading north/east referenced or strictly relative? Thank you.