KumarRobotics/ublox #238: U-blox: received NACK: 0x06 / 0x08 (ZED-F9R)

r4imps, 2024-02-16:
**Describe the bug**
Hi!
I have connected ZED-F9R module to Jetson Orin Nano and as I am trying to launch ublox_gps
 `roslaunch ublox_gps ublox_device.launch param_file_name:=zed_f9p node_name:=gps`
I receive:
[ERROR] [1708075350.434111486]: U-blox: received NACK: 0x06 / 0x08
[FATAL] [1708075350.434316743]: Error configuring u-blox: Failed to set measurement rate to 1000ms and navigation rate to 1
[ERROR] [1708075350.436987007]: U-Blox ASIO input buffer read error: Operation canceled, 0

In config file i have set the device to /dev/ttyACM1 , since I have other devices connected aswell.
and config_on_startup: true
As to rate and nav_rate no matter what I insert I always get the same result.

when setting config_on_startup: false then it works just fine.
baudrate is set to 115200

In addition to that I receive:
[ WARN] [1708075350.426576075]: Product category HPS  from MonVER message not recognized options are HPG REF, HPG ROV, HPG #.#, HDG #.#, TIM, ADR, UDR, LAP, FTS, SPG

My gps module is configured to CFG_RATE_NAV_PRIO = 10 and CFG_NAV_RATE = 4 using pygpsclient. USB and UART1 baudrate set to 115200.

How can I fix the problem to change the nav_rate and rate?

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - ROS Version [e.g. ROS1 Noetic]

closed 2024-02-16 by r4imps

SyedShayaan, 2024-04-01:
Hey, I am facing the same error, would you mind sharing how you got it working? 
@r4imps

r4imps, 2024-04-03:
set the  -> config_on_startup: false in param_file

and in the pygpsclient after configurations there is a setting to save configuration to non-volatile memory (bottom right corner)

thats what i forgot to do everytime i changed my configuration