KumarRobotics/ublox #240: How to get Odometry messages out of this driver ?

chrissunny94, 2024-02-21:
I want the following 

- Sensor_msgs/Odometry  , i am able to compute the Pose from lat, long and altitude .
- I have issues with the Orientation (quaternion )

chrissunny94, 2024-03-08:
Has anyone accumulated pointcloud data from a 3D lidar (such as VLP16) along with the 3DOF Pose from the UBLOX Gnss + IMU fusion .

I want to understand how accurate this was .

tomlogan501, 2024-03-08:
You mean by using a SLAM to compare on a relative level ?

chrissunny94, 2024-03-08:
Yes kind of SLAM , but using the Pointcloud for only building the map (and not for odometry )

The odometry output should be from GNSS/IMU fusion .

chrissunny94, 2024-03-20:
```
#include <chrono>
#include <functional>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
#include "nav_msgs/msg/odometry.hpp"
#include <sensor_msgs/msg/nav_sat_fix.hpp>
#include "sensor_msgs/msg/imu.hpp"

#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#include <gps_tools/conversions.h>
#include <ublox_msgs/msg/hnr_pvt.hpp>
#include <ublox_msgs/msg/nav_att.hpp>

#include "geometry_msgs/msg/transform_stamped.hpp"
#include "tf2/LinearMath/Quaternion.h"
#include "tf2_ros/static_transform_broadcaster.h"

using std::placeholders:: _1;

using namespace std::chrono_literals;

/* This example creates a subclass of Node and uses std::bind() to register a
* member function as a callback from the timer. */

/**
   * Get Geometry (UTM) point from GNSS position, assuming zero origin.
   */
  void UTMPointFromGnss(
          geometry_msgs::msg::Point& pt,
          double lat,
          double lon,
          double hgt)
  {
    pt.z = hgt;

    std::string zone; //unused
    gps_tools::LLtoUTM(lat, lon, pt.y, pt.x, zone);
  }

/***
   * Converts radians to degrees
   *
   * @return degrees
   */
  inline double radiansToDegrees(double radians)
  {
    return radians * 180.0 / M_PI;
  }

  /***
   * Converts degrees to Radians
   *
   * @return radians
   */
  inline double degreesToRadians(double degrees)
  {
    return degrees * M_PI / 180.0;
  }

double latitude , longitude , altitude , roll , pitch , azimuth;
nav_msgs::msg::Odometry odometry;
tf2::Quaternion Z90_DEG_ROTATION; ///< Rotate ENU to ROS frames.
    
class MinimalPublisher : public rclcpp::Node
{
  public:
[... 137 lines cut ...]
      publisher_->publish(message);
      publishOdometry();
    }
    rclcpp::TimerBase::SharedPtr timer_;
    rclcpp::Publisher<std_msgs::msg::String>::SharedPtr publisher_;
    rclcpp::Publisher<nav_msgs::msg::Odometry>::SharedPtr odom_pub_;
    rclcpp::Subscription<ublox_msgs::msg::NavATT>::SharedPtr navatt_sub_;
    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr subscription_;
    rclcpp::Subscription<sensor_msgs::msg::NavSatFix>::SharedPtr navsatfix_subscription_;
    sensor_msgs::msg::Imu imu_;
    std::shared_ptr<tf2_ros::StaticTransformBroadcaster> tf_static_broadcaster_;
    //for Odometry Publisher
    
    size_t count_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalPublisher>());
  rclcpp::shutdown();
  return 0;
}

```