KumarRobotics/ublox #247: Feature request: add protection level message parsing

gennartan, 2024-08-19:
**Describe the solution you'd like**
Forward the protection level data on a specific topic. The output of the protection level is published through the UBX-NAV-PL message. This would be a good way to ensure localization reliability when safety is required. The ublox feature is described in [1].

**Describe alternatives you've considered**
Localization precision is also given by h_acc and v_acc. But it is not sufficient for safety in some applications. We also noticed that those values could be too small compared to real position estimate.

**Additional context**
As described in [1], the protection level feature is designed for robotic applications that requires reliability in the precision of the localization. This feature would therefore be required soon. With this ticket, I hope to accelerate the integration of this feature to be able to use it with my robotic application soon.

[1] https://www.u-blox.com/en/technologies/protection-level

ghanim-mukhtar, 2025-04-29:
Hi @gennartan any updates regarding this issue? Did you manage to implement this ? If so can you share please or make a PR

gennartan, 2025-04-30:
Hello, the PR related to this issue is https://github.com/KumarRobotics/ublox/pull/249

On my feed I also see it just above your comment, so it should be properly linked to this ticket.

I am currently using a local build of ublox using it (and also this one: https://github.com/KumarRobotics/ublox/pull/252) , so I can tell you that it works.