KumarRobotics/ublox #253: GPS lon/lat stucking issue

jmk7791, 2024-10-17:
We are testing with EVKF9P.
When we run the code **roslaunch ublox_gps ublox_device.launch** and check out lon/lat by **rostopic echo /ublox/fix**, a lon/lat stucking issue occurs.

When the position covariance value gets really low (lower than 1.0), sometimes lon/lat and position covariance are fixed to certain values, and do not change even if we move with the GPS.

Is there anyone who suffered from same issue? or how can we deal with it/what will be the reason?

ndmmdn, 2025-01-20:
I do not have this issue, however our relative heading also gets stuck often.

willSapgreen, 2025-08-26:
Hello @jmk7791 
if this hiccup happens when you move the EVKF9P around, not in the stationary condition,
you may want to check if tmode3 is set to 1 which is for the survey-in ( the stationary condition ).
In my case, hiccup was solved after changing tmode3 to 0.
Hope this solves your issue.