KumarRobotics/ublox #254: Getting the error terminate called after throwing an instance of 'std::runtime_error'   what():  Could not configure serial baud rate

RaccoonAD, 2024-10-18:
I don't know why this error appears cause everything seems to be well. I'm using zed-f9p so I set 38400 baud. And when i do roslaunch this appears:

[ INFO] [1729245557.051115804]: U-Blox: Opened serial port /dev/ttyUSB0
[DEBUG] [1729245557.057213218]: Configuring UART1 baud rate: 38400, In/Out Protocol: 7 / 1
[DEBUG] [1729245557.057314877]: Waiting for ACK 0x06 / 0x00
[DEBUG] [1729245557.057484220]: U-Blox sent 28 bytes:
b5 62 06 00 14 00 01 00 00 00 d0 08 00 00 00 96 00 00 07 00 01 00 00 00 00 00 91 84
[DEBUG] [1729245557.057634903]: U-Blox received 41 bytes
00 0a 24 47 4e 52 4d 43 2c 30 39 35 39 31 37 2e 30 30 2c 41 2c 33 37 32 34 2e 37 31 31 35 37 2c 4e 2c 30 30 36 30 30 2e 30
[DEBUG] [1729245557.082188629]: U-Blox received 123 bytes
39 30 36 36 2c 57 2c 30 2e 30 32 31 2c 2c 31 38 31 30 32 34 2c 2c 2c 41 2c 56 2a 30 30 0d 0a 24 47 4e 56 54 47 2c 2c 54 2c 2c 4d 2c 30 2e 30 32 31 2c 4e 2c 30 2e 30 33 39 2c 4b 2c 41 2a 33 34 0d 0a 24 47 4e 47 47 41 2c 30 39 35 39 31 37 2e 30 30 2c 33 37 32 34 2e 37 31 31 35 37 2c 4e 2c 30 30 36 30 30 2e 30 39 30 36 36 2c 57 2c 31 2c 30 36 2c 32 2e 32 32 2c 33 33 2e
[DEBUG] [1729245557.099364062]: U-Blox received 61 bytes
32 2c 4d 2c 34 37 2e 32 2c 4d 2c 2c 2a 36 43 0d 0a 24 47 4e 47 53 41 2c 41 2c 33 2c 30 39 2c 30 34 2c 30 37 2c 33 30 2c 30 36 2c 2c 2c 2c 2c 2c 2c 2c 33 2e 33 35 2c 32 2e 32 32 2c 32
[DEBUG] [1729245557.114753014]: U-Blox received 61 bytes
2e 35 32 2c 31 2a 30 43 0d 0a 24 47 4e 47 53 41 2c 41 2c 33 2c 37 39 2c 36 38 2c 37 38 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 33 2e 33 35 2c 32 2e 32 32 2c 32 2e 35 32 2c 32 2a 30 46 0d 0a 24
[DEBUG] [1729245557.148208506]: U-Blox received 123 bytes
47 4e 47 53 41 2c 41 2c 33 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 33 2e 33 35 2c 32 2e 32 32 2c 32 2e 35 32 2c 33 2a 30 31 0d 0a 24 47 4e 47 53 41 2c 41 2c 33 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 2c 33 2e 33 35 2c 32 2e 32 32 2c 32 2e 35 32 2c 34 2a 30 36 0d 0a 24 47 50 47 53 56 2c 33 2c 31 2c 31 31 2c 30 34 2c 31 34 2c 30 35 35 2c 33 31 2c 30 35 2c 31 30 2c 33 30 31
[DEBUG] [1729245557.164298932]: U-Blox received 61 bytes
2c 2c 30 36 2c 36 31 2c 31 39 36 2c 32 31 2c 30 37 2c 36 30 2c 31 30 38 2c 32 36 2c 31 2a 36 33 0d 0a 24 47 50 47 53 56 2c 33 2c 32 2c 31 31 2c 30 39 2c 34 35 2c 30 34 37 2c 33 32 2c
[DEBUG] [1729245557.178449901]: U-Blox received 61 bytes
31 31 2c 35 35 2c 32 39 34 2c 31 32 2c 31 33 2c 30 39 2c 32 34 30 2c 2c 31 39 2c 30 36 2c 32 30 39 2c 2c 31 2a 36 42 0d 0a 24 47 50 47 53 56 2c 33 2c 33 2c 31 31 2c 32 30 2c 34 30 2c
[DEBUG] [1729245557.212228122]: U-Blox received 123 bytes
33 31 32 2c 31 36 2c 32 39 2c 30 30 2c 33 32 39 2c 2c 33 30 2c 34 38 2c 31 36 37 2c 30 36 2c 31 2a 35 46 0d 0a 24 47 50 47 53 56 2c 33 2c 31 2c 31 31 2c 30 34 2c 31 34 2c 30 35 35 2c 33 30 2c 30 35 2c 31 30 2c 33 30 31 2c 2c 30 36 2c 36 31 2c 31 39 36 2c 32 31 2c 30 37 2c 36 30 2c 31 30 38 2c 2c 36 2a 36 31 0d 0a 24 47 50 47 53 56 2c 33 2c 32 2c 31 31 2c 30 39 2c 34
[DEBUG] [1729245557.225992354]: U-Blox received 61 bytes
35 2c 30 34 37 2c 32 30 2c 31 31 2c 35 35 2c 32 39 34 2c 2c 31 33 2c 30 39 2c 32 34 30 2c 2c 31 39 2c 30 36 2c 32 30 39 2c 2c 36 2a 36 43 0d 0a 24 47 50 47 53 56 2c 33 2c 33 2c 31 31
[DEBUG] [1729245557.241951905]: U-Blox received 61 bytes
2c 32 30 2c 34 30 2c 33 31 32 2c 2c 32 39 2c 30 30 2c 33 32 39 2c 2c 33 30 2c 34 38 2c 31 36 37 2c 32 32 2c 36 2a 35 39 0d 0a 24 47 4c 47 53 56 2c 33 2c 31 2c 31 30 2c 36 37 2c 30 32
[DEBUG] [1729245557.257760107]: U-Blox received 61 bytes
2c 30 36 39 2c 2c 36 38 2c 34 30 2c 30 33 39 2c 31 39 2c 36 39 2c 35 32 2c 33 32 30 2c 2c 37 30 2c 30 35 2c 32 36 39 2c 2c 31 2a 37 39 0d 0a 24 47 4c 47 53 56 2c 33 2c 32 2c 31 30 2c
[DEBUG] [1729245557.289054038]: U-Blox received 119 bytes
37 37 2c 30 38 2c 30 32 34 2c 2c 37 38 2c 33 30 2c 30 38 30 2c 2c 37 39 2c 31 38 2c 31 33 35 2c 2c 38 33 2c 32 37 2c 31 39 32 2c 2c 31 2a 37 36 0d 0a 24 47 4c 47 53 56 2c 33 2c 33 2c 31 30 2c 38 34 2c 35 34 2c 32 38 30 2c 2c 38 35 2c 31 36 2c 33 33 31 2c 2c 31 2a 37 35 0d 0a 24 47 4c 47 53 56 2c 33 2c 31 2c 31 30 2c 36 37 2c 30 32 2c 30 36 39 2c 2c 36 38
[DEBUG] [1729245557.324362465]: U-Blox received 123 bytes
2c 34 30 2c 30 33 39 2c 32 37 2c 36 39 2c 35 32 2c 33 32 30 2c 2c 37 30 2c 30 35 2c 32 36 39 2c 2c 33 2a 37 36 0d 0a 24 47 4c 47 53 56 2c 33 2c 32 2c 31 30 2c 37 37 2c 30 38 2c 30 32 34 2c 2c 37 38 2c 33 30 2c 30 38 30 2c 32 36 2c 37 39 2c 31 38 2c 31 33 35 2c 32 31 2c 38 33 2c 32 37 2c 31 39 32 2c 2c 33 2a 37 33 0d 0a 24 47 4c 47 53 56 2c 33 2c 33 2c 31 30 2c 38 34
[DEBUG] [1729245557.338323887]: U-Blox received 61 bytes
2c 35 34 2c 32 38 30 2c 2c 38 35 2c 31 36 2c 33 33 31 2c 2c 33 2a 37 37 0d 0a 24 47 41 47 53 56 2c 31 2c 31 2c 30 30 2c 37 2a 37 33 0d 0a 24 47 41 47 53 56 2c 31 2c 31 2c 30 30 2c 32
[DEBUG] [1729245557.357676593]: U-Blox received 61 bytes
2a 37 36 0d 0a 24 47 42 47 53 56 2c 31 2c 31 2c 30 30 2c 31 2a 37 36 0d 0a 24 47 42 47 53 56 2c 31 2c 31 2c 30 30 2c 33 2a 37 34 0d 0a 24 47 4e 47 4c 4c 2c 33 37 32 34 2e 37 31 31 35
[DEBUG] [1729245557.368952022]: U-Blox received 36 bytes
37 2c 4e 2c 30 30 36 30 30 2e 30 39 30 36 36 2c 57 2c 30 39 35 39 31 37 2e 30 30 2c 41 2c 41 2a 36 45 0d 0a
[DEBUG] [1729245558.040666066]: U-Blox received 4 bytes
24 47 4e 52
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not configure serial baud rate

bnbhat, 2024-10-18:
Hi @RaccoonAD 

Could you share your config file?

RaccoonAD, 2024-10-18:
Yes, here it is:
# Configuration Settings for ZED-F9P device

debug: 4                              # Range 0-4 (0 means no debug statements will print)

device: /dev/ttyUSB0
#tmode3: 0
frame_id: gps

uart1:
  baudrate: 38400

# Enable u-blox message publishers
publish:
  all: true
  nav:
    all: true
    relposned: true
    posllh: true
    posecef: true

bnbhat, 2024-10-18:
Try setting the `tmode3` as per your usecase.  For more details check this [section](https://github.com/KumarRobotics/ublox#for-hpg-reference-devices) in ReadMe or check this [msg](https://github.com/KumarRobotics/ublox/blob/master/ublox_msgs/msg/CfgTMODE3.msg).

For now, run it with tmode3 set to 0 and let me know if you get the same error.

RaccoonAD, 2024-10-19:
Do you know something else that I can try?

bnbhat, 2024-10-25:
@RaccoonAD does everything work fine with ucenter/ucenter-2?

RaccoonAD, 2024-10-25:
Yep, but I'm currently obtaining another error that says failed to poll mover and set relevant things. But if I use screen on ttyusb0 everythings seems fine. I don't know what to do.

bnbhat, 2024-10-25:
Which version of ROS are you using?

> But if I use screen on ttyusb0 everythings seems fine.

So you are receiving UBX-MON-VER msg when you run `screen`?
MON-VER msg should begin with `0xb5 0x62 0x0a 0x04 .....`

RaccoonAD, 2024-10-25:
16.04

bnbhat, 2024-10-25:
As ROS Kinetic is already end of life, there could be many issues with this driver on that version as it is no longer tested on it.

ghanim-mukhtar, 2025-06-03:
I had a similar problem with `/dev/ttyUSB0` and it turns out to be an issue with baudrate once I used the correct one it worked without any problems.