KumarRobotics/ublox #264: problems when connecting to GPS+XBEE

ghanim-mukhtar, 2025-05-11:
**Describe the bug**
When connecting the device via GPS+XPEE 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to GPS+XBEE
2. Change the device accordingly (in the yaml it should be `/dev/ttyUSB0`
3. launch the node

**Expected behavior**
I expect the node to launch correctly

**Actual behavior**
The node crashes with runtime error
```
[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2025-05-11-13-17-49-462170-delta-ros2-humble-container-cpu-simu-79-464
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ublox_gps_node-1]: process started with pid [465]
[ublox_gps_node-1] [INFO] [1746962269.504652683] [ublox_gps_node]: U-Blox: Opened serial port /dev/ttyUSB0
[ublox_gps_node-1] [DEBUG] [1746962269.505049430] [ublox_gps_node]: Configuring UART1 baud rate: 19200, In/Out Protocol: 32 / 0
[ublox_gps_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ublox_gps_node-1]   what():  Could not configure serial baud rate
[ERROR] [ublox_gps_node-1]: process has died [pid 465, exit code -6, cmd '/home/ros/delta_ws/install/ublox_gps/lib/ublox_gps/ublox_gps_node --ros-args --params-file /home/ros/delta_ws/install/ublox_gps/share/ublox_gps/config/c94_m8t_rover.yaml'].
```

ghanim-mukhtar, 2025-05-13:
I tried to disable the configuration part of the yaml file using `config_on_startup: false` but got the following run time crash:

```
[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2025-05-12-12-45-07-423981-delta-ros2-humble-container-cpu-simu-79-1989
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ublox_gps_node-1]: process started with pid [1990]
[ublox_gps_node-1] [INFO] [1747046707.464252785] [ublox_gps_node]: U-Blox: Opened serial port /dev/ttyUSB0
[ublox_gps_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ublox_gps_node-1]   what():  Failed to poll MonVER & set relevant settings
[ERROR] [ublox_gps_node-1]: process has died [pid 1990, exit code -6, cmd '/home/ros/delta_ws/install/ublox_gps/lib/ublox_gps/ublox_gps_node --ros-args --params-file /home/ros/delta_ws/install/ublox_gps/share/ublox_gps/config/test_config.yaml'].

```

bnbhat, 2025-05-14:
When the `config_on_startup` is set to `false`, no config is set by the driver. So you have to enable the required msgs using `u-center` or [python GUI client](https://github.com/semuconsulting/PyGPSClient). Make sure UBX msgs are enabled.

ghanim-mukhtar, 2025-05-14:
The UBX messages are enabled I believe because when I connect via POWER+GPS everything works fine.

bnbhat, 2025-05-14:
Could you provide more details about the device you are using?

ghanim-mukhtar, 2025-05-14:
I am using the starter kit ZED F9P you can see its details in this [link](https://www.ardusimple.com/product/simplertk2b-heading-basic-starter-kit-ip67/)

bnbhat, 2025-05-15:
Could you please share the debug logs? You can set the debug parameter to 4 (the maximum level), and it should output the actual data being sent.

ghanim-mukhtar, 2025-05-19:
Here is the output with debug at 4:
```
ros@delta-ros2-humble-container-cpu-simu-79:~/delta_ws$ ros2 launch ublox_gps ublox_gps_node-launch.py 
[INFO] [launch]: All log files can be found below /home/ros/.ros/log/2025-05-19-07-58-36-618925-delta-ros2-humble-container-cpu-simu-79-277
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ublox_gps_node-1]: process started with pid [278]
[ublox_gps_node-1] [INFO] [1747634316.722006527] [ublox_gps_node]: U-Blox: Opened serial port /dev/ttyUSB0
[ublox_gps_node-1] [DEBUG] [1747634316.723190765] [ublox_gps_node]: U-Blox sent 8 bytes: 
[ublox_gps_node-1] b5 62 a 4 0 0 e 34 
[ublox_gps_node-1] [DEBUG] [1747634316.737093043] [ublox_gps_node]: U-Blox received 2 bytes 
[ublox_gps_node-1] 0 c0 
[ublox_gps_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ublox_gps_node-1]   what():  Failed to poll MonVER & set relevant settings
[ERROR] [ublox_gps_node-1]: process has died [pid 278, exit code -6, cmd '/home/ros/delta_ws/install/ublox_gps/lib/ublox_gps/ublox_gps_node --ros-args --params-file /home/ros/delta_ws/install/ublox_gps/share/ublox_gps/config/test_config.yaml'].
```

ghanim-mukhtar, 2025-05-19:
I did another test with different configuration file as follows:
```
# Configuration Settings for C94-M8P device
ublox_gps_node:
  ros__parameters:
    debug: 4                    # Range 0-4 (0 means no debug statements will print)

    save:
      mask: 3103                # Save I/O, Message, INF Message, Nav, Receiver
                                # Manager, Antenna, and Logging Configuration
      device: 4                 # Save to EEPROM

    # device: /dev/ttyACM0
    device: /dev/ttyUSB0
    frame_id: gps
    rate: 1.0                   # in Hz
    nav_rate: 1                 # [# of measurement cycles], recommended 1 Hz, may
                                # be either 5 Hz (Dual constellation) or
                                # 8 Hz (GPS only)
    dynamic_model: portable     # Airborne < 2G, 2D fix not supported (3D only),
                                # Max Alt: 50km
                                # Max Horizontal Velocity: 250 m/s,
                                # Max Vertical Velocity: 100 m/s
    fix_mode: auto
    enable_ppp: false           # Not supported by C94-M8P
    dr_limit: 0

    # TMODE3 Config
    tmode3: 0                   # Survey-In Mode
    sv_in:
      reset: false              # True: disables and re-enables survey-in (resets)
                                # False: Disables survey-in only if TMODE3 is
                                # disabled
      min_dur: 300              # Survey-In Minimum Duration [s]
      acc_lim: 3.0              # Survey-In Accuracy Limit [m]

    uart1:
      baudrate: 460800           # C94-M8P specific
      # baudrate: 115200
      in: 32                     # RTCM 3
      out: 32                    # No UART out for rover

    gnss:
      glonass: true             # Supported by C94-M8P
      beidou: false             # Supported by C94-M8P
      qzss: true                # Supported by C94-M8P

    dgnss_mode: 3               # Fixed mode

    inf:
      all: true                 # Whether to display all INF messages in console

    # Enable u-blox message publishers
    publish:
      all: true
      aid:
        hui: false

      nav:
        posecef: false
```

And get the output in the attached text file.

[log_output.txt](https://github.com/user-attachments/files/20280839/log_output.txt)

bnbhat, 2025-05-28:
Hey @ghanim-mukhtar 

Are you connected to UART1 or USB interface?

If UART1, try configuring using any gui like ucenter 2. then run the driver with proper baudrate and config_on_startup set to false. 
This should get you started, later you can configure from this driver.

ghanim-mukhtar, 2025-06-02:
Yes I did that and now it is working with a fix baudrate and for information I am connected to USB (`/dev/ttyUSB0`), thx for the help

closed 2025-06-02 by ghanim-mukhtar